realthunder/FreeCAD #968: [Trojan:Script/Wacatac.B!ml Virus detected]

dingolino, 2024-03-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
Stable 2024.03.17
```

### Subproject(s) affected?

None

### Problem description

I downloaded (both Py 3.8 and 3.11) the Windows install file and my System with MS Defender is telling me that a Virus is detected

Trojan:Script/Wacatac.B!ml

please double check whether this is a false positive.

win11 with all patches

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dingolino, 2024-03-17:
ok, I was able to double chekc on another computer with "eset" Antivirus - no virus was found. Will unpack and check again

Miksu80, 2024-03-20:
Got this same with win11 and MS scanner

dingolino, 2024-03-20:
A new *.7z file was uploaded without the problem.