realthunder/FreeCAD #969: [Problem] No Stable 2024.03.17 build for macOS?

pgilfernandez, 2024-03-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
The last stable release 2024.03.17 does not include a macOS build? was there any problem with it?

Thanks
```

### Subproject(s) affected?

None

### Problem description

Github last Stable 2024.03.17 release doesn't include any macOS build

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pgilfernandez, 2024-03-18:
Maybe I wrote too soon then... let's wait then, thanks for the comment.

oerkel47, 2024-03-18:
Linux Appimage is also still missing.

pgilfernandez, 2024-03-20:
Great, thanks for your comments