realthunder/FreeCAD #970: [Problem] Pocket to datum plane fails with "Null face in SketchBased::getSupportFace()!"

psommerfeld, 2024-03-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 14.0
Word size of FreeCAD: 64-bit
Version: 2024.207.0.14555 (Git shallow)
Build type: Release
Branch: _conda_cache_origin_head
Hash: 0c0f69ad4806682096b51f7669d5f1cb83f9c7d7
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * fasteners.backup1686222008.1392 0.3.50 (Disabled)
  * ProDarkThemePreferencePack 1.0.0
  * BIM 2021.12.0
[/code]
```

### Subproject(s) affected?

None

### Problem description

This works with FreeCad main branch so it can't be OCC issue.
1. Try to create pocket of "sk hex insert" to face datum "datum inside plane" (or any datum plane, it seems)
2. Get this error: <App> Document.cpp(3424): Failed to recompute wire_stripper#Pocket: Null face in SketchBased::getSupportFace()!
[image]

### Anything else?

Same issue with previous Link stage release (2023.05.21).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-03-23:
Could you please post your file here, or email to me?

psommerfeld, 2024-03-26:
[wire stripper.FCStd.zip](https://github.com/realthunder/FreeCAD/files/14760199/wire.stripper.FCStd.zip)
Here is my project

closed 2024-04-30 by realthunder

realthunder, 2024-04-30:
Sorry, missed you comment. The above commit fixed issue. The bug affects sketches that attach to non face object. As a workaround, you can change the the support of `Sketch002` to be the same as `Sketch001`, instead of attaching to it.

psommerfeld, 2024-05-02:
Ok, that's a good enough workaround for now.
Thanks!

commit d2b55a39 (2024-04-30): PD: allow non-face support for sketch based feature

Fixes #970