realthunder/FreeCAD #973: [Problem] Creating subshapebinder causes Link Branch to hang

jbayless, 2024-03-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 2024.319.0.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 20240317stable)
Hash: 07c85fedeae51f76d1999f15b24d3c921f623f8d
Python 3.11.8, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: English/Canada (en_CA)
Installed mods: 
  * lattice2 1.0.0
  * Manipulator
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

If I create a document with two bodies, draw a sketch in the first body, and then in the second body create a SubShapeBinder referencing the sketch in the first body, it immediately causes FreeCAD to freeze indefinitely. This process worked fine in the 20240123 release.

I've attached a simple test file showing this problem. "smile.FCstd" was created in 20240317. Try making "Face1" the active body, select the smile sketch from the Layout body, and create a SubShapeBinder. On my computer at least, FreeCAD will hang forever at this point. You can also open "smile2.FCstd", which I created in 20240123. It has the SubShapeBinder already created; however, if you right click each body and choose "Recompute Object", FreeCAD will freeze when Face1 is recomputed.

[freecad_smile.zip](https://github.com/realthunder/FreeCAD/files/14704556/freecad_smile.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-03-21:
Fixed by be6b793036e. This is serious problem. I'll make a new release soon.

closed 2024-03-21 by realthunder

vincentbaeten, 2024-03-22:
Thanks. I had the same problem and it seems to be fixed now.