realthunder/FreeCAD #976: What are the long-term intentions of this fork?

eblanshey, 2024-03-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
N/A
```

### Subproject(s) affected?

None

### Problem description

Hi RealThunder, thank you for all the innovations in this fork! Besides the toponaming fix, there are many awesome experimental features here, such as selecting sketch areas, padding with tolerances, new preferences, etc. I have questions about your intentions:

1. What is your goal with continuing putting these features in this branch? Do you put them here because getting them merged into the main repo is too slow? 
2. Is there, or will there be a process for getting these features merged into the main branch? Or is it just "someone who wants it will open a PR in the main repo?" What do you think? (For now I put [one request](https://github.com/FreeCAD/FreeCAD/issues/13128) in the main branch.)
3. Do you envision continuing with this fork indefinitely to experiment, or will you move development to the main repo after a certain point, such as the full toponaming merge?
4. Can you comment on future intentions to make files in this branch compatible with main branch or not?

I am asking these because which version to use (this one vs Ondsel with upcoming Assembly workbench vs main branch) is difficult to choose, so knowing intentions here can help with the decision. Other people may also like to know this, and possibly help you if your goals are clear.

I am posting this here since I didn't see these questions answered in the forums. Hope you don't mind!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-03-24:
> 1. What is your goal with continuing putting these features in this branch? Do you put them here because getting them merged into the main repo is too slow?

Yes, that's true. Upstream lacks man power to review my changes. And many new features on my branch are either big or incremental, meaning that it is built on top of some thing big, like the topo naming.

> 2\. Is there, or will there be a process for getting these features merged into the main branch? Or is it just "someone who wants it will open a PR in the main repo?" What do you think? (For now I put [one request](https://github.com/FreeCAD/FreeCAD/issues/13128) in the main branch.)

Sure there is. Some of my big PRs have already gotten merged. But some must wait, for example your request here due to the fact that it is written on top of my topo naming code. Basically the whole PartDesign modeling code is refactored on top of it. 

> 3\. Do you envision continuing with this fork indefinitely to experiment, or will you move development to the main repo after a certain point, such as the full toponaming merge?

I will most likely continue the current development pattern as I find it easier to manage both for me and other developers. Topo naming code are intensively reviewed and merged by other developers right now. A big chunk of the code is already in the upstream. It's just that they want to fully understand the code before merging it. So the process is slow. I'd say they've already done over half of it.

> 4\. Can you comment on future intentions to make files in this branch compatible with main branch or not?

I will keep backward compatibility to upstream, meaning that you should be able to open files from upstream in my branch, but apparently not the other way round, because it is out of my control. Many features do not exist in upstream.

eblanshey, 2024-03-25:
Thank you very much for your clarifications, much appreciated. After the topo naming merge I hope things start moving faster in the main branch to bring everything to parity faster. Cheers!

closed 2024-03-25 by eblanshey