realthunder/FreeCAD #978: Problems in the Pie Menu

thecoolsundar, 2024-03-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2024.322.0.38974 (Git) Snap 130
Build type: Release
Branch: tag: 20240322stable
Hash: da98001cfe5cc50cb2e9e4c241626d07dd4b06f5
Python 3.10.12, Qt 5.15.10, Coin 4.0.1rt, Vtk 7.1.1, OCC 7.7.1
Locale: English/India (en_IN)
Installed mods: 
  * CurvedShapes 1.0.8
  * AirPlaneDesign 0.4.0
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

I mapped 'Leave Sketch'  to X shortcut key in the pie menu using customize 

When I try to exit the sketch using the pie menu, FreeCAD crashes

Also I mapped 'create sketch' in the Part Design work bench to C shortcut key in the pie menu

When I select the pie menu Icon before the plane, it works normally but when I select the plane first and then the pie menu icon, FreeCAD crashes

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-03-31:
Could you please launch snap from command line. If you can reproduce the crash, there will be call stack dump printed to the terminal. Please post the dump here.

thecoolsundar, 2024-03-31:
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7b3c17ee5520]
#1  0x7b3c19b93118 in QMenu::setActiveAction(QAction*) from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x18
#2  0x7b3c1afbf1e6 in StdCmdToolbarSubMenu::activated(int) from /snap/freecad-realthunder/130/usr/lib/libFreeCADGui.so+0x36
#3  0x7b3c1afe829b in Gui::Command::_invoke(int, bool) from /snap/freecad-realthunder/130/usr/lib/libFreeCADGui.so+0x26b
#4  0x7b3c1afe8786 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /snap/freecad-realthunder/130/usr/lib/libFreeCADGui.so+0x156
#5  /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f48f4) [0x7b3c18ea18f4]
#6  0x7b3c199fcc26 in QAction::triggered(bool) from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x46
#7  0x7b3c199ff91c in QAction::activate(QAction::ActionEvent) from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xac
#8  0x7b3c1b05d76d in Gui::ShortcutManager::onTimer() from /snap/freecad-realthunder/130/usr/lib/libFreeCADGui.so+0x1ed
#9  /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f48f4) [0x7b3c18ea18f4]
#10  0x7b3c18ea597e in QTimer::timeout(QTimer::QPrivateSignal) from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x3e
#11  0x7b3c18e972bf in QObject::event(QEvent*) from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x1cf
#12  0x7b3c19a03763 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#13  0x7b3c1af62bd8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad-realthunder/130/usr/lib/libFreeCADGui.so+0x98
#14  0x7b3c18e69e4a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#15  0x7b3c18ec266b in QTimerInfoList::activateTimers() from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x3db
#16  /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x315fb4) [0x7b3c18ec2fb4]
#17  /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x7b3c15f95d3b]
#18  /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab258) [0x7b3c15feb258]
#19  /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7b3c15f933e3]
#20  0x7b3c18ec3338 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#21  0x7b3c18e6876b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#22  0x7b3c18e70d14 in QCoreApplication::exec() from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#23  0x7b3c1ae8e876 in Gui::Application::runApplication() from /snap/freecad-realthunder/130/usr/lib/libFreeCADGui.so+0x426
#24  /snap/freecad-realthunder/130/usr/bin/FreeCADLink(+0x9649) [0x5f1cb6a0a649]
#25  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7b3c17eccd90]
#26  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7b3c17ecce40]
#27  /snap/freecad-realthunder/130/usr/bin/FreeCADLink(+0x9a75) [0x5f1cb6a0aa75]

thecoolsundar, 2024-03-31:
I hope this is what you meant. I am new to freecad and to linux.thanks

thecoolsundar, 2024-03-31:
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7bdbfc416520]
#1  0x7bdbfe0c4118 in QMenu::setActiveAction(QAction*) from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x18
#2  0x7bdbff4f01e6 in StdCmdToolbarSubMenu::activated(int) from /snap/freecad-realthunder/130/usr/lib/libFreeCADGui.so+0x36
#3  0x7bdbff51929b in Gui::Command::_invoke(int, bool) from /snap/freecad-realthunder/130/usr/lib/libFreeCADGui.so+0x26b
#4  0x7bdbff519786 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /snap/freecad-realthunder/130/usr/lib/libFreeCADGui.so+0x156
#5  /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f48f4) [0x7bdbfd3d28f4]
#6  0x7bdbfdf2dc26 in QAction::triggered(bool) from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x46
#7  0x7bdbfdf3091c in QAction::activate(QAction::ActionEvent) from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xac
#8  0x7bdbff58e76d in Gui::ShortcutManager::onTimer() from /snap/freecad-realthunder/130/usr/lib/libFreeCADGui.so+0x1ed
#9  /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f48f4) [0x7bdbfd3d28f4]
#10  0x7bdbfd3d697e in QTimer::timeout(QTimer::QPrivateSignal) from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x3e
#11  0x7bdbfd3c82bf in QObject::event(QEvent*) from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x1cf
#12  0x7bdbfdf34763 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#13  0x7bdbff493bd8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad-realthunder/130/usr/lib/libFreeCADGui.so+0x98
#14  0x7bdbfd39ae4a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#15  0x7bdbfd3f366b in QTimerInfoList::activateTimers() from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x3db
#16  /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x315fb4) [0x7bdbfd3f3fb4]
#17  /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x7bdbfa4c7d3b]
#18  /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab258) [0x7bdbfa51d258]
#19  /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7bdbfa4c53e3]
#20  0x7bdbfd3f4338 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#21  0x7bdbfd39976b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#22  0x7bdbfd3a1d14 in QCoreApplication::exec() from /snap/freecad-realthunder/130/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#23  0x7bdbff3bf876 in Gui::Application::runApplication() from /snap/freecad-realthunder/130/usr/lib/libFreeCADGui.so+0x426
#24  /snap/freecad-realthunder/130/usr/bin/FreeCADLink(+0x9649) [0x5dd9b8c5e649]
#25  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7bdbfc3fdd90]
#26  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7bdbfc3fde40]
#27  /snap/freecad-realthunder/130/usr/bin/FreeCADLink(+0x9a75) [0x5dd9b8c5ea75]

thecoolsundar, 2024-03-31:
The first dump is when I attempt to enter sketcher by selecting the plane BEFORE clicking the sketcher Icon in the pie menu. 

The second dump is the crash that occurs when I attempt to exit sketcher using the pie menu.

realthunder, 2024-03-31:
Could you please post your user settings here `Tools -> Edit parameters -> Export`?

thecoolsundar, 2024-03-31:
<?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<FCParameters>

  <FCParamGroup Name="Root">
    <FCParamGroup Name="BaseApp">
      <FCParamGroup Name="LogLevels">
        <FCInt Name="Default" Value="2"/>
      </FCParamGroup>
      <FCParamGroup Name="Preferences">
        <FCParamGroup Name="General">
          <FCParamGroup Name="HiddenPreferencePacks"/>
          <FCParamGroup Name="Widgets">
            <FCParamGroup Name="Gui::Dialog::DlgPreferencesImp">
              <FCText Name="Geometry">489 27 1131 454</FCText>
            </FCParamGroup>
            <FCParamGroup Name="Gui::Dialog::DlgCustomizeImp">
              <FCText Name="Geometry">618 27 1113 660</FCText>
            </FCParamGroup>
            <FCParamGroup Name="Gui::Dialog::DlgCustomKeyboardImp">
              <FCText Name="splitter">486 575 </FCText>
            </FCParamGroup>
            <FCParamGroup Name="Gui::Dialog::DlgCustomToolbars">
              <FCText Name="splitterVertical">341 197 </FCText>
              <FCText Name="splitter">456 609 </FCText>
            </FCParamGroup>
            <FCParamGroup Name="Gui::Dialog::DlgIconBrowser"/>
            <FCParamGroup Name="Gui::Dialog::DlgCustomizeSpaceball"/>
            <FCParamGroup Name="Gui::Dialog::DlgParameterImp">
              <FCText Name="Geometry">220 140 800 600</FCText>
              <FCText Name="splitter3">256 518 </FCText>
              <FCBool Name="Maximized" Value="0"/>
            </FCParamGroup>
            <FCParamGroup Name="TaskFilletParameters">
              <FCUInt Name="ColumnSize4" Value="191"/>
            </FCParamGroup>
            <FCParamGroup Name="FileDialog">
              <FCText Name="Geometry">308 197 624 412</FCText>
            </FCParamGroup>
            <FCParamGroup Name="Gui::DlgObjectSelection">
              <FCText Name="Geometry">598 347 795 375</FCText>
              <FCText Name="Splitter">142 142 </FCText>
              <FCText Name="VSplitter">769 0 </FCText>
            </FCParamGroup>
            <FCParamGroup Name="TaskChamferParameters"/>
          </FCParamGroup>
          <FCText Name="FileOpenSavePath">/home/theboss</FCText>
          <FCText Name="Language">English</FCText>
          <FCText Name="LastModule">PartDesignWorkbench</FCText>
          <FCBool Name="AutoApplyPreference" Value="1"/>
          <FCBool Name="SaveUserParameter" Value="1"/>
          <FCBool Name="SubstituteDecimalSeparator" Value="0"/>
          <FCBool Name="EnableCursorBlinking" Value="1"/>
          <FCBool Name="ShowSplasher" Value="1"/>
          <FCBool Name="ConfirmAll" Value="0"/>
          <FCInt Name="ToolbarIconSize" Value="16"/>
          <FCInt Name="UseLocaleFormatting" Value="0"/>
          <FCInt Name="WorkbenchTabIconSize" Value="0"/>
          <FCInt Name="ToolTipIconSize" Value="64"/>
        </FCParamGroup>
        <FCParamGroup Name="Units">
          <FCInt Name="UserSchema" Value="0"/>
          <FCInt Name="Decimals" Value="2"/>
          <FCInt Name="FracInch" Value="8"/>
        </FCParamGroup>
        <FCParamGroup Name="Macro">
          <FCText Name="MacroPath">/home/theboss/snap/freecad-realthunder/common</FCText>
          <FCText Name="ScriptFile">FullScript.FCScript</FCText>
          <FCBool Name="LocalEnvironment" Value="1"/>
          <FCBool Name="RecordGui" Value="1"/>
          <FCBool Name="GuiAsComment" Value="1"/>
[... 1394 lines cut ...]
        </FCParamGroup>
        <FCParamGroup Name="SketcherAngle">
          <FCText Name="Hist0">125 °</FCText>
          <FCText Name="Hist1">120.00 °</FCText>
          <FCText Name="Hist2">120 °</FCText>
          <FCText Name="Hist3">130.00 °</FCText>
          <FCText Name="Hist4">130 °</FCText>
        </FCParamGroup>
      </FCParamGroup>
      <FCParamGroup Name="CheckMessages">
        <FCBool Name="Delete" Value="1"/>
      </FCParamGroup>
    </FCParamGroup>
    <FCParamGroup Name="Tux">
      <FCParamGroup Name="NavigationIndicator">
        <FCBool Name="Compact" Value="0"/>
        <FCBool Name="Tooltip" Value="1"/>
      </FCParamGroup>
      <FCParamGroup Name="PersistentToolbars">
        <FCInt Name="Deprecated" Value="2"/>
      </FCParamGroup>
    </FCParamGroup>
  </FCParamGroup>

</FCParameters>

thecoolsundar, 2024-03-31:
Sorry. it generates an xml file but I can't upload that. when I paste the contents here, it becomes like above

thecoolsundar, 2024-03-31:
[parameters.zip](https://github.com/realthunder/FreeCAD/files/14814031/parameters.zip)

thecoolsundar, 2024-03-31:
Hope that helps

realthunder, 2024-03-31:
You user config does not have any pie menu configured. Could you please create your pie menu and then post the settings again? I did find and fixed a crash related to shortcut timer. But the call stack here is a bit different from yours.

thecoolsundar, 2024-03-31:
[image]
[image]

thecoolsundar, 2024-03-31:
I have lots of pie menus and they are all working. not sure why it's not showing in configurations. I created the menus from Tools->customize -> toolbars

realthunder, 2024-03-31:
I see. I thought you're creating global pie menu.

thecoolsundar, 2024-03-31:
[parameters.zip](https://github.com/realthunder/FreeCAD/files/14814302/parameters.zip)

thecoolsundar, 2024-03-31:
[image]

thecoolsundar, 2024-03-31:
I'm sorry for the misunderstanding. I'm still figuring out FreeCAD. And I love your branch.

The pie menus really help to speed up my workflow

realthunder, 2024-03-31:
Thanks for your report. Should be fixed by the above commit. It seems you are among the very few to explore the workbench contextual Pie Menu, or else this bug won't be hiding for so long.

closed 2024-03-31 by realthunder

thecoolsundar, 2024-03-31:
thank you Sir! how do I get this to update on my computer?

realthunder, 2024-03-31:
I will make a new release next week.

thecoolsundar, 2024-04-04:
can you please tell me how to update my software on ubuntu once these changes are made? thanks

realthunder, 2024-04-04:
Snap will auto update at background, once you close the app. To update manually, run `sudo snap refresh freecad-realthunder`.

commit 21933672 (2024-03-31): Gui: fix crash on contextual pie menu switching workbench

Fixes #978