realthunder/FreeCAD #983: FreeCAD freezes entering wrong data in feature "pad with two dimensions"[Problem]

wsteffe, 2024-04-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu-wayland)
Word size of FreeCAD: 64-bit
Version: 0.21.0.38986 (Git)
Build type: Unknown
Branch: LinkStable
Hash: 21933672e35d6f4646d6a80e5ffb432ee658b517
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.0
Locale: English/United States (en_US)
Installed mods: 
  * asm3 0.12.0
  * Assembly4 0.50.2
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

More specifically it freezes if lenght2 < - lenght1. This is a wrong condition because it leads to a negative total lenght (total lenght=lenght1+lenght2).
But the problem is that if such wrong dimensions are entered (by error), it is necessary to kill FC and all work done is lost brcause recovering data are not avalilable.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-04-06:
What are the exact length1 and length2 value you tried. I can't seem to reproduce the problem.

wsteffe, 2024-04-09:
Hi RT,
 I also can't reproduce anymore. It happened with a deleted document. In the meantime I have also resynced my local repo so I do not know if it has been solved in the last commits or it was specific to a particular FC document.
I think it is better to close it for now,  I will let you know it it happens again. Thanks for attantion.

closed 2024-04-09 by wsteffe