realthunder/FreeCAD #989: [Problem] Problem when extruding circles "Up to face"

kotovasia5120, 2024-04-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 2024.407.0.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 20240407stable)
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.11.8, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * fasteners 0.5.15
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

In the new version, now none of the circles are extruded.
Although the problem was marked as  resolved. [(https://github.com/realthunder/FreeCAD/issues/985)]
1:at a distance
[image]

2:To the edge.
[image]
```
20:30:10  <Part> ViewProviderExt.cpp(2239): Cannot compute Inventor representation for the shape of _9#Pocket001: Bnd_Box is void
20:30:18  <Part> ViewProviderExt.cpp(2239): Cannot compute Inventor representation for the shape of _9#Pocket001: Bnd_Box is void
20:30:27  <Part> ViewProviderExt.cpp(2239): Cannot compute Inventor representation for the shape of _9#Pocket001: Bnd_Box is void
20:30:27  <Part> ViewProviderExt.cpp(2239): Cannot compute Inventor representation for the shape of _9#Pocket001: Bnd_Box is void
20:31:01  <Part> ViewProviderExt.cpp(2239): Cannot compute Inventor representation for the shape of _9#Pocket001: Bnd_Box is void
20:31:05  <Part> ViewProviderExt.cpp(2239): Cannot compute Inventor representation for the shape of _9#Pocket001: Bnd_Box is void
20:31:05  <Part> ViewProviderExt.cpp(2239): Cannot compute Inventor representation for the shape of _9#Pocket001: Bnd_Box is void
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-04-09:
I accidentally packaged wrong version of OCC. I have since triggered re-packaging. Please download the release image again and try.

kotovasia5120, 2024-04-09:
> I accidentally packaged wrong version of OCC. I have since triggered re-packaging. Please download the release image again and try.
Now extrusion works through, and not to the edge.
[image]
```
20:07:52  <App> PropertyLinks.cpp(392): _4#Sketch003.AttachmentSupport auto change element reference _4#Pocket001 ;#11b;:M#11a;RFI;:Hb4e:b,F.Face15 -> ;#1a3;:M#11a;RFI;:Hb4e:b,F.Face15
20:07:52  <App> PropertyLinks.cpp(392): _4#Pocket002.UpToFace auto change element reference _4#Pocket001 ;#ef;:Hb4e,F.Face7 -> ;#ef;:M;CUT;:Hb4e:7,F.Face7
```
```
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 2024.408.0.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 20240407stable)
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.11.8, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * fasteners 0.5.15
  * Help 1.0.3
[/code]

```

kotovasia5120, 2024-04-09:
Same thing on the other version.
```
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 2024.409.0.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 20240407stable)
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.8.19, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * fasteners 0.5.15
  * Help 1.0.3
[/code]

```
[4.zip](https://github.com/realthunder/FreeCAD/files/14922016/4.zip)

wohltat, 2024-08-04:
I experience the same problem. "up to face", "up to first" and "up to last" are not working.

----------------------------
```
OS: Manjaro Linux (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 2024.409.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20240407stable)
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: English/United States (en_US)
```

wohltat, 2024-08-12:
> I accidentally packaged wrong version of OCC. I have since triggered re-packaging. Please download the release image again and try.

I tried the release image, but still bug still exists.
Would it help to rebuild the latest version myself?

realthunder, 2024-08-12:
Sorry for overlooking this problem. I'll fix it soon and make a new release.

commit ac3496a1 (2024-08-16): Part: fix extrude/pad until face

Fixes #989
commit 8e0c802a (2024-08-16): Part: fix extrude/pad until face

Fixes #989
closed 2024-08-16 by realthunder