realthunder/FreeCAD #994: [Problem] Transformation Offset moves original feature as well

oerkel47, 2024-04-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 2024.409.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20240407stable)
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

Using Mirror/Linear pattern (I only tested those two) with transform offsets moves not only the new feature but also the original one. 
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-04-20:
The `TransformOffset` is meant to be applied to the original shape, because otherwise there is no easy way to achieve the same effect. If you only want to transform the the `Transformed` feature, you can simply move the mirror plane.

realthunder, 2024-04-20:
Just elaborate a bit more In case it is not obvious. Although the preview shows both the unmoved and moved original sub-feature plus the mirrored one, if you exit the editing dialog, the unmoved original will disappear. This is because the preview by default shows the whole feature in the previous history step plus the new sub-feature. You can turn off preview and see the final shape if that's confusing. I think I can work on that a bit in the future to avoid confusion.

The special thing about `TransformOffset` is that it can move the original sub-feature in the previous history step. If that's all the user wants, I mean moving the previous sub-feature without adding new sub-feature, he can use a `LinearPattern` with pattern count set to one. Note that this `Special` functionality of `TransformOffset` only applies to the immediate historical sub-feature. If you select any sub-feature before that, there will be warning message issued and `TransformOffset` will only be applied to the new sub-feature.  See screen cast below.

https://github.com/realthunder/FreeCAD/assets/1207888/28519c26-6e21-454f-9742-7ad0c4cddfb0

oerkel47, 2024-04-20:
Thank you for the elaboration. I am pretty sure that I have used this before to only move the new mirrored feature in a certain direction which I think can be really handy. Simply moving the mirror plane does not give me the option to offset in x,y,z.

From a user pov I think it is not a very logical thing to do it like this within the mirror or linear pattern functionality. If I wanted the original and new features somewhere else I would just edit the original feature before the mirror/pattern. I now understand this behaviour comes from the fact that the original feature is treated as part of the pattern and thus receives the modification as well. Is it possible to have a checkbox "move only new subfeature" kind of like the "workaround warning behaviour" you described?

realthunder, 2024-09-28:
[image]

commit ec878c9a (2024-09-28): PD: add OffsetBaseFeature property to Transformed feature

For explicit control whether to apply TransformOffset to base feature

Closes #994
commit 8428dfee (2024-09-28): PD: add OffsetBaseFeature property to Transformed feature

For explicit control whether to apply TransformOffset to base feature

Closes #994
closed 2024-09-28 by realthunder