realthunder/FreeCAD #995: [Problem] PartDesign Pad UpToFace broken if face to be padded exceeds the target

twelho, 2024-04-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33771 (Git)
Build type: Release
Branch: LinkStable
Hash: 9335dd6ad7b53a0c2b93ea84df51361c83eb2219
Python 3.11.9, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.6.21
  * Assembly4 0.50.6
  * CfdOF 1.24.9
```

### Subproject(s) affected?

PartDesign

### Problem description

When padding a face in the `UpToFace` mode, if the boundaries of the face to be padded exceed the face to be padded up to, FreeCAD `LinkStable` breaks with the following error message:

```
Document.cpp(3424): Failed to recompute Test#Pad001: BRepFeat_MakePrism: extrusion failed
```

[image]

When trying the same with `LinkMerge`, the pad works as intended:

[image]

### Anything else?

Here is a simple [MRE](https://en.wikipedia.org/wiki/Minimal_reproducible_example):

[Test.zip](https://github.com/realthunder/FreeCAD/files/15051727/Test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-04-21:
Please try the flatpak build in my release. I packaged a patched version of OCC the CAD kernel library. In next release, I'll try improve compatibility with the official OCC.

twelho, 2024-04-21:
Thanks for the quick response, the Flatpak from https://github.com/realthunder/FreeCAD/releases/tag/20240407stable works with the given example. The about page for it states OCC 7.7.2 (patched or not), while [my Flatpak](https://github.com/twelho/org.freecadweb.FreeCAD/) still has OCC 7.6.3, so I'll need to update that first to see if it fixes the issue. I presume your patched OCC lies [here](https://github.com/realthunder/OCCT), so I can also test that until upstream compatibility is improved.

closed 2024-04-21 by twelho