realthunder/FreeCAD #1001: [Problem] toolbar items become stuck if you accidentally drop them in the top row next to "file" "edit" etc.

kevenwyld, 2024-05-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu Core 22 (sway/sway)
Word size of FreeCAD: 64-bit
Version: 2024.409.0.38991 (Git) Snap 132
Build type: Release
Branch: tag: 20240407stable
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.10.12, Qt 5.15.10, Coin 4.0.1rt, Vtk 7.1.1, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly3 0.12.2
  * fasteners 0.5.10
```

### Subproject(s) affected?

None

### Problem description

When moving toolbars around if you accidentally drop them in the top row the "grab bar" dotted vertical lines disappear and you can't move them again.  So they get stuck there.

Here's a screenshot, you can see that the toolbars below the top position have the dotted grab area on the left but the ones at the top do not.
[image]

Thanks!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kevenwyld, 2024-05-05:
I fixed it by moving the toolbars listed in link.user.cfg

from
```
<FCParamGroup Name="MenuBarLeft"/>
```

to

```
<FCParamGroup Name="ToolBars">
    <FCParamGroup Name="Movable">
```

and changing the type from `FCInt` to `FCBool`.  That put them back to their default locations and I was able to move them again, this time being careful to avoid the top bar.

Here's a diff, though it's a little hard to follow without the whole context:

```
***************
*** 1286,1296 ****
            <FCBool Name="Path Modification" Value="1"/>
          </FCParamGroup>
          <FCParamGroup Name="MenuBarRight"/>
!         <FCParamGroup Name="MenuBarLeft">
!           <FCInt Name="Structure" Value="0"/>
!           <FCInt Name="Sketcher edit tools" Value="1"/>
!           <FCInt Name="Sketcher geometries" Value="2"/>
!         </FCParamGroup>
          <FCParamGroup Name="ToolBars">
            <FCParamGroup Name="Movable">
              <FCBool Name="*" Value="1"/>
--- 1286,1292 ----
            <FCBool Name="Path Modification" Value="1"/>
          </FCParamGroup>
          <FCParamGroup Name="MenuBarRight"/>
!         <FCParamGroup Name="MenuBarLeft"/>
          <FCParamGroup Name="ToolBars">
            <FCParamGroup Name="Movable">
              <FCBool Name="*" Value="1"/>
***************
*** 1318,1323 ****
--- 1314,1322 ----
              <FCBool Name="Sketcher tools" Value="1"/>
              <FCBool Name="Sketcher virtual space" Value="1"/>
              <FCBool Name="Structure" Value="1"/>
+             <FCBool Name="Sketcher edit tools" Value="1"/>
+             <FCBool Name="Sketcher geometries" Value="1"/>
+             <FCBool Name="Structure" Value="1"/>
              <FCBool Name="View" Value="1"/>
              <FCBool Name="Workbench" Value="1"/>
            </FCParamGroup>
***************
```

realthunder, 2024-05-06:
Right click an active tool button and select menu item Undock.

kevenwyld, 2024-05-06:
> Right click an active tool button and select menu item Undock.

Thanks!  That worked.  Sorry to bother you for nothing.

closed 2024-05-06 by kevenwyld