realthunder/FreeCAD #1002: [Problem] PartDesign Mirrored

free777cat, 2024-05-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Linux Mint 22 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 2024.409.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20240407stable)
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * FEMbyGEN 2.1.0
  * Sketcher
  * frame 0.1.1
  * Assembly4 0.50.6
  * mbdyn-dynamics
  * sheetmetal_середа, 10 квітня 2024.zip
  * fasteners 0.4.72
  * sheetmetal_субота, 06 квітня 2024.zip
  * sheetmetal.zip
  * lattice2 1.0.0
  * FEM_FrontISTR 0.1.0
  * sheetmetal_work_dxf.zip
  * sheetmetal_субота, 30 грудня 2023.zip
  * SheetMetalFoldCmd.py_повністю_робоча_неділя, 21 квітня 2024.zip
  * DynamicData 2.56.0
  * sheetmetal 0.4.13
  * Defeaturing 1.2.2
  * dxf-library
  * OpticsWorkbench 1.0.12
  * temp
  * ExplodedAssembly
[/code]
```

### Subproject(s) affected?

None

### Problem description

PartDesign Mirrored
It is impossible to leave only the mirrored part.

Only the mirrored part is needed for further actions.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-09-28:
[image]

[image]

Please note that `HideBaseFeature` option will hide all previous features and leave only the new transformed feature, so it will produce the final result as shown below.

[image]

closed 2024-09-28 by realthunder

free777cat, 2024-09-29:
will this be pushed to the main FreeCAD branch?

commit fc7a6be1 (2024-09-28): PD: Add HideBaseFeature to Transformed Feature

Closes #1002
commit cdab8655 (2024-09-29): PD: Add HideBaseFeature to Transformed Feature

Closes #1002