realthunder/FreeCAD #1019: Many errors opening FC document after last commit

wsteffe, 2024-07-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.38998 (Git)
Build type: Unknown
Branch: (HEAD detached at 0e9cf4aa7c)
Hash: 0e9cf4aa7c7bac94a83edd07b4bd74a71a26446c
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.0
Locale: English/United States (en_US)
Installed mods: 
  * asm3 0.12.0
  * Assembly4 0.50.2
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

Hi RT,  please open the annexd FC file with current LinkStable. You will get a lot of errors. These problems disappear switching back to commit 0e9cf4aa7c7bac94a83edd07b4bd74a71a26446c

### Anything else?

[dielResonFilter_6_0_tunable.zip](https://github.com/user-attachments/files/16280758/dielResonFilter_6_0_tunable.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-07-19:
Could you please post detail logs about the error? I cannot reproduce the problem with my local build.

wsteffe, 2024-07-19:
The file has changed in the  meantime so here I am uploading last version (which generated the reported errors) together with the cache file (extension FCStd1) because I think that the errors come from there.

[dielResonFilter_6_0_tunable.zip](https://github.com/user-attachments/files/16311011/dielResonFilter_6_0_tunable.zip)

Here are the logs:

```
12:51:32  Cannot create object 'Pad': (Cannot add static property)
12:51:32  Cannot create object 'Extrusion': (Cannot add static property)
12:51:32  Cannot create object 'Pad010': (Cannot add static property)
12:51:32  Cannot create object 'Pad011': (Cannot add static property)
12:51:32  Cannot create object 'Pad012': (Cannot add static property)
12:51:32  Cannot create object 'Pad017': (Cannot add static property)
12:51:32  Cannot create object 'Pad018': (Cannot add static property)
12:51:32  Cannot create object 'Pad019': (Cannot add static property)
12:51:32  Cannot create object 'Pad020': (Cannot add static property)
12:51:32  Cannot create object 'Pocket': (Cannot add static property)
12:51:32  Cannot create object 'Pocket001': (Cannot add static property)
12:51:32  Cannot create object 'Pocket002': (Cannot add static property)
12:51:32  Cannot create object 'BaseFeature': (Cannot add static property)
12:51:32  Cannot create object 'Fillet': (Cannot add static property)
12:51:32  Cannot create object 'Fillet001': (Cannot add static property)
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Pad while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Pad017 while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Pocket while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Pocket001 while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Pocket002 while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Fillet while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Fillet001 while loading, maybe an object was not loaded correctly
12:51:32  Lost link to 'Fillet001' while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Pad while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Pad017 while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Pocket while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Pocket001 while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Pocket002 while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Fillet while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Fillet001 while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Pad010 while loading, maybe an object was not loaded correctly
12:51:32  Lost link to 'Pad010' while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Pad010 while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Pad011 while loading, maybe an object was not loaded correctly
12:51:32  Lost link to 'Pad011' while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Pad011 while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Pad012 while loading, maybe an object was not loaded correctly
12:51:32  Lost link to 'Pad012' while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Pad012 while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable BaseFeature while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Extrusion while loading, maybe an object was not loaded correctly
12:51:32  Lost link to 'BaseFeature' while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable BaseFeature while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(1031): Lost link to dielResonFilter_6_0_tunable Extrusion while loading, maybe an object was not loaded correctly
12:51:32  <App> PropertyLinks.cpp(4291): Lost link to 'Pad011' while loading, maybe an object was not loaded correctly
[... cut ...]
12:51:32  Lost link to 'Pad017' while loading, maybe an object was not loaded correctly
12:51:32  Lost link to 'Pocket' while loading, maybe an object was not loaded correctly
12:51:32  Lost link to 'Pocket' while loading, maybe an object was not loaded correctly
12:51:32  3e-08 <App> Document.cpp(2452): dielResonFilter_6_0_tunable#Binder018.Support: Link not restored
Object: Pad011
12:51:32  0.0190972 <App> Document.cpp(2452): dielResonFilter_6_0_tunable#Binder019.Support: Link not restored
Object: Pad012

```

wsteffe, 2024-07-19:
I have just tried to remove the file with extension .FCStd1 but the errors come out anyway.

wsteffe, 2024-08-12:
Same errors also with LinkMerge
```
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.41224 (Git)
Build type: Unknown
Branch: LinkMerge
Hash: 0fd396ddbec0e760a0eeb96f9b4548c28b167c07
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.0
Locale: English/United States (en_US)
Installed mods: 
  * asm3 0.12.0
  * Assembly4 0.50.2
```

```
FreeCAD 0.22.0, Libs: 0.22.0devR41224 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Loading Assembly workbench...
During initialization the error "'NoneType' object has no attribute 'text'" occurred in freecad.asm3
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 237, in InitApplications
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/walter/.local/share/FreeCAD/Mod/asm3/./freecad/asm3/init_gui.py", line 3, in <module>
    from . import gui
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/home/walter/.local/share/FreeCAD/Mod/asm3/./freecad/asm3/gui.py", line 10, in <module>
    from .utils import getElementPos,objName,addIconToFCAD,guilogger as logger
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/home/walter/.local/share/FreeCAD/Mod/asm3/./freecad/asm3/utils.py", line 10, in <module>
    import FreeCAD, FreeCADGui, Part, Draft
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/usr/local/Mod/Draft/Draft.py", line 53, in <module>
    from draftutils.utils import ARROW_TYPES as arrowtypes
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/usr/local/Mod/Draft/draftutils/utils.py", line 42, in <module>
    from draftutils import params
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/usr/local/Mod/Draft/draftutils/params.py", line 513, in <module>
    PARAM_DICT = _get_param_dictionary()
  File "/usr/local/Mod/Draft/draftutils/params.py", line 474, in _get_param_dictionary
    path, entry, value = _param_from_PrefCheckBox(widget)
  File "/usr/local/Mod/Draft/draftutils/params.py", line 226, in _param_from_PrefCheckBox
    path = elem.find("cstring").text
AttributeError: 'NoneType' object has no attribute 'text'
--------------------------------------------------------------------------------
Cannot create object 'Pad': (Cannot add static property)
[... 45 lines cut ...]
Lost link to 'Pad021' while loading, maybe an object was not loaded correctly
<App> PropertyLinks.cpp(1026): Lost link to dielResonFilter_6_0_tunable_v6 Pad021 while loading, maybe an object was not loaded correctly
<App> PropertyLinks.cpp(4279): Lost link to 'Pad011' while loading, maybe an object was not loaded correctly
<App> PropertyLinks.cpp(4279): Lost link to 'Pad012' while loading, maybe an object was not loaded correctly
<App> PropertyLinks.cpp(1026): Lost link to dielResonFilter_6_0_tunable_v6 Pad018 while loading, maybe an object was not loaded correctly
Lost link to 'Pad018' while loading, maybe an object was not loaded correctly
<App> PropertyLinks.cpp(1026): Lost link to dielResonFilter_6_0_tunable_v6 Pad018 while loading, maybe an object was not loaded correctly
<App> PropertyLinks.cpp(1026): Lost link to dielResonFilter_6_0_tunable_v6 Pad019 while loading, maybe an object was not loaded correctly
Lost link to 'Pad019' while loading, maybe an object was not loaded correctly
<App> PropertyLinks.cpp(1026): Lost link to dielResonFilter_6_0_tunable_v6 Pad019 while loading, maybe an object was not loaded correctly
<App> PropertyLinks.cpp(1026): Lost link to dielResonFilter_6_0_tunable_v6 Pad020 while loading, maybe an object was not loaded correctly
Lost link to 'Pad020' while loading, maybe an object was not loaded correctly
<App> PropertyLinks.cpp(1026): Lost link to dielResonFilter_6_0_tunable_v6 Pad020 while loading, maybe an object was not loaded correctly
Lost link to 'Pad' while loading, maybe an object was not loaded correctly
[... cut ...]
Lost link to 'Pad017' while loading, maybe an object was not loaded correctly
Lost link to 'Pad017' while loading, maybe an object was not loaded correctly
Lost link to 'Pocket' while loading, maybe an object was not loaded correctly
Lost link to 'Pocket' while loading, maybe an object was not loaded correctly
4.3e-08 <App> Document.cpp(2476): dielResonFilter_6_0_tunable_v6#Binder018.Support: Link not restored
Object: Pad011
0.0192568 <App> Document.cpp(2476): dielResonFilter_6_0_tunable_v6#Binder019.Support: Link not restored
Object: Pad012

```

realthunder, 2024-08-16:
Cloud you please remove the entire build directory and do a clean build and see?

wsteffe, 2024-08-22:
Hello RT, I made a clean build/install from a new build directory but nothing changed.
I also made a preference/reset but even that didn't solve the problem.
The last FreeCAD versions (stable and linkmerge) don't work with any of my old files.

wsteffe, 2024-08-22:
I have tried to use Freecad-realthunder from snap repository.
It works but it is older then my last working linkedge build:

- Current Snap Version: 38991 (Git) Snap 132
- My last working build: 38998 (Git)

wsteffe, 2024-08-22:
The problem disappered after rebuilding LinkMerge in debug mode.
I have also observed that nothing was set for CMEKE_BUILD_TYPE before entering "debug", which is strange because a defalult value should always be set.
So I do not know which CMEKE_BUILD_TYPE was used in the previous build.

wsteffe, 2024-08-23:
The problem rappeares when I set CMEKE_BUILD_TYPE=release.
But now I have fixed it (in LinkMerge) removing the lines 523 and 525 in src/App/PropertyContainer.cpp b/src/App/PropertyContainer.cpp.
The change assures same behaviour in debug/release mode of the function:

```
void PropertyData::addProperty(OffsetBase offsetBase,const char* PropName, Property *Prop, const char* PropertyGroup , PropertyType Type, const char* PropertyDocu)
  {
 -#ifdef FC_DEBUG
      if(!parentMerged)
 -#endif
      {
          short offset = offsetBase.getOffsetTo(Prop);
          if(offset < 0)
```

xplwowi, 2024-08-24:
I have the same problem. The solution from above that @wsteffe suggested solves it.

closed 2024-08-25 by realthunder

wsteffe, 2024-08-26:
The problem is fixed but why this kind of problem has to be detected only with a relaese and not with a debug build ?

```
void PropertyData::addProperty(OffsetBase offsetBase,const char* PropName, Property *Prop, const char* PropertyGroup , PropertyType Type, const char* PropertyDocu)
{
#ifdef FC_DEBUG
    if(!parentMerged)
#endif
    {
        short offset = offsetBase.getOffsetTo(Prop);
        if(offset < 0)
            throw Base::RuntimeError("Invalid static property");
        auto &index = propertyData.get<1>();
        auto it = index.find(PropName);
        if(it == index.end()) {
            if(parentMerged)
                throw Base::RuntimeError("Cannot add static property");
            index.emplace(PropName, PropertyGroup, PropertyDocu, offset, Type);
        } else{
#ifdef FC_DEBUG
            if(it->Offset != offset) {
                FC_ERR("Duplicate property '" << PropName << "'");
            }
#endif
        }
    }

```
When FC_DEBUG is defined, the following line will never be reached
because it should hold (!parentMerged &&  parentMerged).
 
` throw Base::RuntimeError("Cannot add static property");
`

commit a6d29d4b (2024-08-25): Part: fix regression on file loading

Fixes #1019
commit 239a7bf3 (2024-08-25): Part: fix regression on file loading

Fixes #1019