realthunder/FreeCAD #1022: [Problem] Diameter dimension text is cut off

jeffmakes, 2024-07-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Fedora Linux 40 (Workstation Edition) (sway)
Word size of FreeCAD: 64-bit
Version: 2024.409.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20240407stable)
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.50.5
[/code]
```

### Subproject(s) affected?

None

### Problem description

Hi

When a diameter dimension is applied to a circle or arc, the text label is mostly cut off. Radius constraints work fine. Behaviour is the same for driven and reference constraints:
[image]

Cheers
Jeff

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jeffmakes, 2024-08-10:
Please see attached. Thanks.

[param.FCParam.txt](https://github.com/user-attachments/files/16570464/param.FCParam.txt)

xplwowi, 2024-08-24:
This is Linux font issue. Already solved in upstream by using different diameter symbol. Look at my report here: #196 and simple upstream solution here: https://github.com/realthunder/FreeCAD/commit/ad408a0989e90c9a1bc19337b29a8b4ec0f6c611