realthunder/FreeCAD #1025: [Problem] AdditiveLoft generates invalid geometry (not an issue in mainline 0.21.2)

mh-dm, 2024-08-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 20.04.6 LTS (Unity:Unity7:ubuntu/unity)
Word size of FreeCAD: 64-bit
Version: 2024.409.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20240407stable)
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

Here's a simple file using an additive loft that's supposed to result in a tube-like shape (remove .zip extension):
[broken-additive-loft.FCStd.zip](https://github.com/user-attachments/files/16615223/broken-additive-loft.FCStd.zip)

This is the invalid geometry:
[image]

The issue is more clearly seen in a slicer:
[image]

The two sketches used have at least 2 axes of symmetry so the expectation is that the the resulting additive loft geometry keeps said symmetry.

### Anything else?

This issue does not happen in the latest mainline FreeCAD 0.21.2 release.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-08-16:
What version of mainline FreeCAD are you using. I tried the latest Linux AppImage release and it has the same problem.

mh-dm, 2024-08-16:
Using the latest released version https://github.com/FreeCAD/FreeCAD/releases/tag/0.21.2
[code]
OS: Ubuntu 20.04.6 LTS (Unity:Unity7:ubuntu/unity)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]

To have it working you just have to reopen the file then delete the AdditiveLoft and recreate it. I'm attaching a working version:
[working-in-21.2-additive-loft.FCStd.zip](https://github.com/user-attachments/files/16637731/working-in-21.2-additive-loft.FCStd.zip)

If you however open this file in Link version it will ask to recompute which will break the geometry.

mh-dm, 2024-08-16:
The mainline 0.21.2 version I was using was from freecad stable PPA.
```
$ apt info freecad-python3
Package: freecad-python3
Version: 2:0.21.2+dfsg1~202403021917~ubuntu20.04.1
```

To make sure that wasn't a confounding factor, I've downloaded the latest mainline Linux AppImage release and tried it. The behaviour was exactly the same as the 0.21.2 from PPA. It just loads way slower because it's an AppImage and AppImage is bloated and slow. I'm a bit surprised that version info is identical
```
OS: Ubuntu 20.04.6 LTS (Unity:Unity7:ubuntu/unity)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
```

realthunder, 2024-08-16:
This is most likely a problem of the newer OCCT 7.7. Lastest mainline FreeCAD 0.22 also uses this version which has the same problem. I'll try to find some workaround.