realthunder/FreeCAD #1027: [Problem] Can't stretch to the surface of the cylinder

3xxx, 2024-08-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 2024.408.0.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 20240407stable)
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.11.8, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: Chinese/China (zh_CN)
Installed mods: 
  * A2plus 0.4.64
  * Assembly3 0.12.2
  * Assembly4 0.50.9
  * BillOfMaterials 0.0.7
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

The model I drew in 2023 with the link version, using up to face, it was okay to , but now this version is not working.
The official weekly build does. The left side of the picture below is link, and the right side is the official weekly compilation
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-08-18:
Related to #989 

Fixed in 8e0c802a6d

3xxx, 2024-08-19:
Can you help compile a version for us to use in the near future? Thank you.

realthunder, 2024-08-19:
Yes, I do plan to.