realthunder/FreeCAD #1028: [Problem] Missing external geometry reference

3xxx, 2024-08-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 2024.408.0.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 20240407stable)
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.11.8, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: Chinese/China (zh_CN)
Installed mods: 
  * A2plus 0.4.64
  * Assembly3 0.12.2
  * Assembly4 0.50.9
  * BillOfMaterials 0.0.7
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

The following parametric model, when changing the parameters, it will lead to the loss of reference lines, please help to see what the reason is, in our modeling process, a large number of this situation, I don't know whether it is a problem with modeling skills or software algorithms. Thank.
[missing reference.zip](https://github.com/user-attachments/files/16713709/missing.reference.zip)
[image]
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-08-24:
As you may have known, my branch has the new topo naming feature that can track geometry change, but only to some extent. There will always be corner cases. And here is a typical one. In one of your configurations, you refers to the following edge,
[image]

However, when you change to the other configuration, that edge is split into two. The current algorithm cannot guess which one the user intend to refer. It would be dangerous to just pick any one without notifying user, as it may cause subtle error that are hard to pinpoint in later stage of modeling.
[image]

The solution in your particular case is easy. Refer the geometry in `Pad` instead of `Pad001`. The principle here is to always select a geometry reference as early as possible in model history since it is less likely to change.

3xxx, 2024-08-24:
Thank you for taking the time to solve the problem for me. The edge you said became 2, I did find out, and I found that its naming changed, and the original reason is here, and I didn't think about it at the time.  Thanks again and good luck.

realthunder, 2024-08-26:
In case of corner cases where the aforementioned workaround fails, the above commit allows the user to explicitly specify the geometry reference without topo name tracking. Here is how it works. You add a new column in the configuration table with simple text geometry reference.
[image]

The purple background is because I added an alias to the cell named `reference`. Now, create a `SubShapeBinder` and binds to that geometry. Then adds an expression to its `Support` property by right click the property in the property editor and choose `Expression...`, and enter the following expression.
[image]

Finally, change the sketch external reference to the binder. For existing external reference, you can use `Attach geometry` command. Here is the critical part. Instead of selecting the geometry in 3D view, you need to hold `Alt` key and select the `Binder` object in the tree view. This will set the reference to the entire object instead of a sub-element, which effectively disables topo name tracking, so that the sketch won't report error when switching configuration.
[image]

[image]

3xxx, 2024-10-04:
Thank you once again for your careful guidance.