realthunder/FreeCAD #1029: [Problem] fillet edges lost after a simple dimensional (not topological) change.

wsteffe, 2024-08-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.41236 (Git)
Build type: Release
Branch: LinkMerge
Hash: a6d27c43c5cc399eec64f81ddd7c8dec784a51c6
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.0
Locale: English/United States (en_US)
Installed mods: 
  * asm3 0.12.0
  * Assembly4 0.50.2
```
```

### Subproject(s) affected?

PartDesign

### Problem description

Using the annexed file inside the parameters spreadsheet change cavity_h=12 to something else. Fillet and Fillet001 (inside the body DIEL_CAVITY) get broken with several missing edges.
Only the height of DIEL_CAVITY was changed whiles the topology is invariant. It is very strange that TPN fails in this simple case. 

### Anything else?

[dielResonFilter_6_0_tunable_v6.zip](https://github.com/user-attachments/files/16797326/dielResonFilter_6_0_tunable_v6.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-08-29:
You project is created using an early version of topo naming algorithm. Although the code change between versions are very small, but the produced names can be quite different, and there is a risk of failure in name tracking like what you reported here. I used to explicitly warn user about version changes by popping up a window asking for full recompute. But then thought it is too intrusive. 

Anyway, to fix your problem, you must perform a full recompute before making any changes to the model. Right click the document item in the tree view, and select `Mark to recompute`, and then click `Recompute`. This will do a full recompute. Later change of the parameter shall work without problem.

wsteffe, 2024-08-29:
Hi RT, but have you tried to make a recompute on the annexed document and then change the parameter ? 
It seems to me that it doesn't fix the problem.

realthunder, 2024-08-31:
Yes, I have tried it. It fails on your side? Which parameter did you change and to what value?

wsteffe, 2024-08-31:
Yes, in example it fails when I set <parameters><cavity_h=13 but actually it fails any time for any change. Edges are lost in the Fillet selected in the second picture. After I fix it (reselecting again all missing edges) a second problem pops up in the next fillet (Fillet001). A full recompute doesn't help. 
I should not be using an old version of the naming algo. The project was created about 40 days ago with the last LinkStable (last at that time). Now I am working with current Linkmerge but I have tried also with current LinkStable and it fails in the same way.

[image]
[image]

realthunder, 2024-08-31:
Looks like another Release vs Debug build problem. I'll take a deeper look.

commit 516a9eff (2024-09-03): Part: disable randomization of duplication element resolving

Fixes #1029
commit bb96bd26 (2024-09-03): Part: disable randomization of duplication element resolving

Fixes #1029
closed 2024-09-03 by realthunder