realthunder/FreeCAD #1037: [Problem] Should I wait for AppImage?

free777cat, 2024-10-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu Core 22 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 2024.1003.0.39043 (Git) Snap 133
Build type: Release
Branch: tag: 20241003stable
Hash: 5a84e3dbfa165fbcb00ebd464943d981bb8edf7a
Python 3.10.12, Qt 5.15.10, Coin 4.0.1rt, Vtk 7.1.1, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * sheetmetal 0.4.26
[/code]
```

### Subproject(s) affected?

None

### Problem description

Should I wait for AppImage?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-10-04:
I am trying to fix the build right now. Probably needs a few more days.

3xxx, 2024-10-04:
I am waiting for AppImage and win.

3xxx, 2024-10-04:
Will the new link version introduce the better features in the official version, such as the huge update in the sketch?
Thank you.

closed 2024-10-06 by free777cat