realthunder/FreeCAD #1038: [Problem] FreeCAD crashed on touch input

plundell, 2024-10-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu Core 22 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 2024.1003.0.39043 (Git) Snap 133
Build type: Release
Branch: tag: 20241003stable
Hash: 5a84e3dbfa165fbcb00ebd464943d981bb8edf7a
Python 3.10.12, Qt 5.15.10, Coin 4.0.1rt, Vtk 7.1.1, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Problem description

**Reproduction:**
1. Connect touchscreen to laptop and ensure it works. 
2. Start FreeCAD. 
3. Touch anywhere inside FreeCAD window and **FreeCAD crashes** with this error:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7fa48ca58520]
#1  0x7fa4815bb446 in QXcbConnection::xi2ProcessTouch(void*, QXcbWindow*) from /snap/freecad-realthunder/133/kf5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/../../../libQt5XcbQpa.so.5+0x46
#2  0x7fa4815867ad in QXcbConnection::handleXcbEvent(xcb_generic_event_t*) from /snap/freecad-realthunder/133/kf5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/../../../libQt5XcbQpa.so.5+0x1bd
#3  0x7fa481587d96 in QXcbConnection::processXcbEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad-realthunder/133/kf5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/../../../libQt5XcbQpa.so.5+0x76
#4  /snap/freecad-realthunder/133/kf5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/../../../libQt5XcbQpa.so.5(+0x76f57) [0x7fa4815aff57]
#5  /snap/freecad-realthunder/133/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x7fa48ab03d3b]
#6  /snap/freecad-realthunder/133/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab258) [0x7fa48ab59258]
#7  /snap/freecad-realthunder/133/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7fa48ab013e3]
#8  0x7fa48da44338 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad-realthunder/133/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#9  0x7fa48d9e976b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad-realthunder/133/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#10  0x7fa48d9f1d14 in QCoreApplication::exec() from /snap/freecad-realthunder/133/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#11  0x7fa48fa10316 in Gui::Application::runApplication() from /snap/freecad-realthunder/133/usr/lib/libFreeCADGui.so+0x426
#12  /snap/freecad-realthunder/133/usr/bin/FreeCADLink(+0x9649) [0x5593ddeac649]
#13  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7fa48ca3fd90]
#14  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7fa48ca3fe40]
#15  /snap/freecad-realthunder/133/usr/bin/FreeCADLink(+0x9a75) [0x5593ddeaca75]
```

### Anything else?

**Laptop**: Lenovo ThinkPad T490s 20NX003UMH
**OS**: Linux Mint 20.1
**Kernel**: 5.18.19-051819-generic
**Touchscreen**: Lenovo ThinkVision m14t (USB C)

I have tried on main branch FreeCAD 0.19, 0.20, 0.21 and 1.0 RC2 with the same results. This is the error output from RC2:
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f4812642520]
#1  0x7f480d0e08cf in QXcbConnection::xi2ProcessTouch(void*, QXcbWindow*) from /tmp/.mount_freeca6W6ahl/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5+0x2f
#2  0x7f480d0af5f3 in QXcbConnection::handleXcbEvent(xcb_generic_event_t*) from /tmp/.mount_freeca6W6ahl/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5+0x183
#3  0x7f480d0b0a96 in QXcbConnection::processXcbEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freeca6W6ahl/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5+0x66
#4  /tmp/.mount_freeca6W6ahl/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64be3) [0x7f480d0d5be3]
#5  /tmp/.mount_freeca6W6ahl/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7f4810f10ad7]
#6  /tmp/.mount_freeca6W6ahl/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7f4810f13ac7]
#7  /tmp/.mount_freeca6W6ahl/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f4810f140ce]
#8  0x7f4812d0aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freeca6W6ahl/usr/bin/../lib/libQt5Core.so.5+0x66
#9  0x7f4812cb570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freeca6W6ahl/usr/bin/../lib/libQt5Core.so.5+0x11b
#10  0x7f4812cbd212 in QCoreApplication::exec() from /tmp/.mount_freeca6W6ahl/usr/bin/../lib/libQt5Core.so.5+0x82
#11  0x7f48158c84e1 in Gui::Application::runApplication() from /tmp/.mount_freeca6W6ahl/usr/bin/../lib/libFreeCADGui.so+0xa41
#12  /tmp/.mount_freeca6W6ahl/usr/bin/freecad(+0x789f) [0x55852b2b289f]
#13  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f4812629d90]
#14  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f4812629e40]
#15  /tmp/.mount_freeca6W6ahl/usr/bin/freecad(+0x7c12) [0x55852b2b2c12]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-17:
@realthunder any ideas on a diagnosis here ?

realthunder, 2024-10-24:
Looks like it crashed inside Qt. Looks similar to https://bugreports.qt.io/browse/QTBUG-92054