realthunder/FreeCAD #1042: [Problem] Sketcher.  Auxiliary geometry. (additional panel)

free777cat, 2024-10-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Linux Mint 22 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: 8058dcf52c2661edabf36d8147d5b8a187378d2d
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)

[/code]
```

### Subproject(s) affected?

None

### Problem description

Sketcher .    Is it possible to create an additional panel in Sketcher?
Auxiliary geometry. It should always be enabled in this state.
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

free777cat, 2024-10-29:
The presence of an additional panel will significantly speed up the work process.

3xxx, 2024-10-29:
The sketcher itself is a plane, what is the purpose of re-establishing an "identical" plane? Can you give an example?
If you want to create a plane of another space, it is impossible to support it in Sketcher.

free777cat, 2024-10-30:
The idea is that there is no need to switch between "blue" and "white" geometry. We will have two separate panels that can be placed differently for convenience.

3xxx, 2024-10-30:
Oh, I see what you mean, I understand it as a work plane.