realthunder/FreeCAD #1048: [Backport] FreeCAD 1.0

CharlemagneLasse, 2024-11-19:
### Original PR

https://github.com/FreeCAD/FreeCAD/releases/tag/1.0.0

### Description and Justification

FreeCAD 1.0 is a significant improved version of FreeCAD which can either  be used as new based for the realthunder fork. Or the changes in the realthunder fork could be ported to the the upstream FreeCAD version

wohltat, 2024-11-19:
I'm also very much interested to see both forks come together. 
FC 1.0 has some great improvements in the GUI like dimensioning, and on the other hand this fork (realthunder) is still my favorite because of features like multi part bodies and autogenerate internal geometries. 

Which is more likely, that RT features will be integrated into FC official or the other way round? 

How can one help with that?

CharlemagneLasse, 2024-11-21:
> multi part bodies

Also work with freecad 1.0. You just have to select in the body properties "Allow compound". And this is at least with my config the default.

> and autogenerate internal geometries.

You mean from a sketch - yes, this is really nice. You can have something similar (for "master sketches") by selecting all the lines belonging to a internal geometry. But yes, the automatically generated internal geometries make this easier to use.

What I am also missing in FreeCAD upstream are:

* PartDesign Split tool
*  Sketches allow to use external geometry as defining geometry
*  hovering over a feature in PartDesign tree view highlights the feature in 3d view
*  preview of body which is used for cutting when doing pockets/holes/.... (because I never know in which direction a pocket is done and then wonder why nothing gets cut)

kevenwyld, 2024-11-23:
> hovering over a feature in PartDesign tree view highlights the feature in 3d view

That feature preview is so very nice with complex models.

Also the preview improvements on fillets and chamfurs are quite good, and the fact that you can click to add an edge while also seeing the preview rather than having to toggle back and forth.  Part design pipe operations have the same efficiency improvement.  And fillets also work slightly better in this fork than in upstream, even the dev branch.

In the past I wouldn't have added to a thread like this as it's not really how issues on GH are suppose to work, but these days I think it's important to say out loud the positive things we are thinking.  And I do really appreciate all the things this fork and RT does, both in improving upstream and also creating cool new features and improvements.

vespaman, 2024-12-07:
I just tried 1.0 for a couple of days, and can only agree 100%. In fact, being without;

> hovering over a feature in PartDesign tree view highlights the feature in 3d view

 
> Also the preview improvements on fillets and chamfurs are quite good, and the fact that you can click to add an edge while also seeing the preview rather than having to toggle back and forth. Part [...]

> preview of body which is used for cutting when doing pockets/holes/.... (because I never know in which direction a pocket is done and then wonder why nothing gets cut)

 .. is probably impossible for me. So I'm back on link again. 1.0 has a lot of nice features though, so it is with regret.

twelho, 2025-01-04:
Another observation is that you still can't really reorder features in a body in upstream 1.0. For example, trying to move a pad from after a bunch of fillets to before them causes the fillets to break. Then you'd need to do lots of manual cleanups. In LinkMerge, you can just drag to reorder and it figures it out fully automatically most of the time. So this fork is definitely still needed.

drwho495, 2025-03-09:
The sketcher is significantly easier to use in 1.0, also there is now an assembly workbench (which isnt very stable, so merging isn't really necessary) Im going back to this fork because the TNP mitigation is still way better, its a lot faster, and is more stable.

julian-weinert, 2025-03-25:
To me 1.0 is a total regression in so many ways.
And another one to add to your folks list: Additive and subtractive features have no translucent preview in mainline.
You cannot see the feature until you accept and create it.

eblanshey, 2025-03-25:
As much as I love the RT fork and its enhancements, it's just sad to see hard work being split into two incompatible versions. RT initiated the efforts due to solving the TNP + an Assembly workbench. Now that V1 already has both, it'd be wonderful to have RT working on adding new features into the main branch. I wish someone would hire him to dedicate work to backport all his improvements, and then continue his efforts with the rest of the dev community on main.