realthunder/FreeCAD #1049: Suggestion to create separate project

julian-weinert, 2024-11-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS Mojave (10.14)
Word size of FreeCAD: 64-bit
Version: 2024.409.0.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 20240407stable)
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: C/Default (C)
Installed mods: 
  * Curves.backup1719938211.117405 0.6.5 (Disabled)
  * A2plus.backup1719938210.597776 0.4.60g (Disabled)
  * Assembly4.backup1719938210.954688 0.12.5 (Disabled)
  * ProDarkThemePreferencePack 1.0.0
  * freecad.gears 1.0.0
  * fasteners.backup1719938211.346364 0.4.52 (Disabled)
  * BIM021 2021.12.0
  * TabBar
  * library
  * Manipulator.backup1719938211.687624 1.4.9 (Disabled)
  * Assembly3.backup1719938210.780728 0.11.4 (Disabled)
  * kicadStepUpMod.backup1719938211.529082 10.16.8 (Disabled)
  * CADExchanger
  * fasteners 0.5.28
  * lattice2 1.0.0
  * Estimate 0.1.2
  * ExtremeProDark 2.6.4
  * Mechatronic
  * ToolbarStyle
  * ExplodedAssembly
  * Curves 0.6.46
[/code]
```

### Subproject(s) affected?

None

### Problem description

This is not a problem, but since discussion forums aren't available in this repo I'm forced to open an issue.

Unfortunately, after the release of FreeCAD 1.0 I had to realise that the main branch has many shortcomings compared to the link branch. I was trying to transition back to main to realise I can't even hunt down those features in old 0.21 versions.

So with this I'm suggesting to convert the link branch into a separate project, going a different direction.
This branch has always been working much better, did far more progress in very short times than main.
I hate to say it, but main isn't going in a direction I'm personally happy with and the response by the main branch team to me trying to figure out missing / broken features was basically "Well, it's risky to use a branch by one guy, so meh"

Would be interesting to see if this is something anyone else would appreciate or be interested in. The open source CAD world has been sadly neglected and having a second project, introducing some competition, could really elevate this community.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Jiiua, 2024-11-26:
I agree, i have been trying to use official FreeCAD and there is always some new-old issue that cycles back upstream, so much so that i had to downgrade a couple times or use this fork, which has been suiting quite well specially since the 1.0 release that has set it back in many ways.

wohltat, 2024-11-26:
Seems similar to #1048 .

I don't get why there is the need for another separate project when the problem is to bring to forks closer to each other.

julian-weinert, 2025-03-25:
@wohltat separating from FreeCAD main would allow independent development and fixes where mainline has regressions. It would also totally remove the need to bring forks closer to each other.
Looking at the performance so far I don't see a bright future for mainline.
There is a reason for Ondsel and AstroCAD to exist... But I think people with the actual real DNA of FreeCAD under their belt would do much better.
Which realthunder has proven well IMHO