realthunder/FreeCAD #1055: is it this project still alive ?

wsteffe, 2025-01-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
Last linkstable and linkmerge.
```

### Subproject(s) affected?

None

### Problem description

Hi RT, a long time has passed since the last commit. I am wondering if you have stopped this project or you are still going to make new developments such as, in example, the realignment with current upstream version.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

col-b, 2025-01-09:
I agree as this branch is still my preferred, even after the 1.0 release, but there are some pretty nice features on the mainline as well. I have some software experience (even if not a ton of free time) and would love to get involved and maybe help keep this alive! Is there a good entry point where one could help get involved?

realthunder, 2025-01-10:
It's not abbandened. I am still adding new things and fixing bugs. It's just that I got busy at work since last Nov and expect to get some break at the end of this month. The LinkMerge branch has already been merged with upstream 2024.05, which is pretty close to 1.0. I will try to make a release this month.

dingolino, 2025-01-10:
Thank you for your feedback!

3xxx, 2025-01-22:
Really looking forward to it, thanks for the reply!

6rzegorz, 2025-01-28:
When can we expect the Link branch to be merged with the upstream 1.0/1.1 version?

ultramakinist, 2025-01-29:
FreeCAD Link3 has not been surpassed in terms of philosophy and
functionality yet. I am also looking forward to the version combined with
the upper version.

6rzegorz ***@***.***>, 28 Oca 2025 Sal, 23:05 tarihinde şunu
yazdı:

glambx, 2025-01-29:
Realthunder's branch is what saved freecad for me, personally.  I'd spent about a month working on a fairly complex project a few years ago and mainline was making me just miserable.  When I discovered rt's branch I announced to everyone watching me suffer "his fork of freecad is as good as mainline is bad!"

Faster, works the way it should, fewer bugs, nicer interface.. even still today 1.0 has a bunch of bugs I can't get past (ie. dragging the view with both mouse buttons in Blender mode).

I don't understand why realthunder's branch doesn't just become the official one.  It's so good!

6rzegorz, 2025-01-29:
glambx > I don't understand why realthunder's branch doesn't just become the official one. It's so good!
....

I agree 100%. The main branch is getting better and is slowly catching up with RT, but it still has a long way to go. 
They have directly adopted some solutions (e.g. the toponaming problem). 
Instead of that, they should make RT as the main branch and develop it together, instead of reinventing the wheel.

herrgerd, 2025-02-01:
> Realthunder's branch is what saved freecad for me, personally. I'd spent about a month working on a fairly complex project a few years ago and mainline was making me just miserable. When I [...]

I agree. There's a few features in official 1.0 which I really miss in RT's branch:

- Linear patterns in which you can set the distance between the features
- sketcher has a bunch of features I miss, especially the offset tool.
- they integrated a finally usable working assembly mode (the ondsel solver)

I do really wish for a combination of both worlds. Only RT's branch does the toponaming in a good way!!!

LinuksGuru, 2025-03-03:
Hi, Realthunder, when you expect to release new version? Waiting it impatiently... Thank!

realthunder, 2025-03-04:
I have started the building process. It may take several days, because some of the CI process needs update and so on.

LinuksGuru, 2025-03-21:
realthunder, do you have plans to release flatpack version of 2025.03? AppImage crashes on my SuSE, can't fix it for now, previous flatpak runs just fine. Thanks in advance.

realthunder, 2025-03-22:
> realthunder, do you have plans to release flatpack version of 2025.03? AppImage crashes on my SuSE, can't fix it for now, previous flatpak runs just fine. Thanks in advance.

I tried but is currently failing. I'll try again soon.

3xxx, 2025-05-18:
Hi @realthunder , may I inquire if the new version of the plan has been made available? I am eagerly looking forward to reviewing it.

3xxx, 2025-06-28:
Hi @realthunder
Hey, what's the latest upgrade plan? Really looking forward to it!

realthunder, 2025-08-16:
> I mean the following in the nicest way possible:
>
> I think it would be a good idea to tell the people in some kind of "pinned
> to the top" post what the current state of the project is. If there won't
> be an update in nearly a year I'd tell my community about it. It's not
> about reasons or justifying yourself, just making sure people are not
> waiting for updates which won't come.
>
> Thanks for all your work you out into making FreeCAD what it is today!

Sorry to keep everyone waiting. There are actually a few fixes since the
last release in March. In case you missed it, it is here,
https://github.com/realthunder/FreeCAD/releases/tag/Tip. It's a big merge
from the upstream, so there are quite some issues reported form users, some
of which I have fixed, but just got no time to make a release. I am so busy
these days. I will try my best to squeeze some time for a release in the
near future.

dingolino, 2025-08-16:
> Sorry to keep everyone waiting. There are actually a few fixes since the
> last release in March. In case you missed it, it is here,
> https://github.com/realthunder/FreeCAD/releases/tag/Tip. It's a big [...]

How to get there???
Whenever I was looking at Github I have only seen this: (attached screen shot)

[image]

dingolino, 2025-08-17:
> Trying this Version and after start I am getting this warning: libpng warning: iCCP: duplicate
This was solved by removing all installed Makros (some of them wherequite old)

> 
> When I open a recent project and open a sketch I am getting this error:
> 
> 18:22:01 Show.ShowUtils.is3DObject error: No SWIG wrapped library loaded

This problem is remaining , even so so when I start with a fresh unmodified FreeCAD

> 
> ```
> OS: Windows 11 build 26100
> Word size of FreeCAD: 64-bit
> Version: 2025.316.0dev.Unknown
> Build type: Release
> Python 3.11.8, Qt 5.15.15, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
> Locale: German/Germany (de_DE)
> Installed mods: 
>   * btl 0.9.9
>   * Defeaturing 1.2.6
>   * MeshRemodel 1.10.37
>   * MeshRemodel.backup1748687648.5592425 1.9.23 (Disabled)
>   * offline-documentation
>   * offline-documentation.backup1748687652.3044467 1.0.0-alpha (Disabled)
> ```

realthunder, 2025-10-05:
I will definitely make a new release this year. But most likely not based on 1.1. I know there's quite a few exciting new features in 1.1, but the code base has diverged so much, it is practically impossible to merge from my side. However, it might be more feasible to pick things you want and add to 1.1 code base. So for everyone still interested, please list the things you like, and I will see if I can add.

drwho495, 2025-10-05:
> I will definitely make a new release this year. But most likely not based on 1.1. I know there's quite a few exciting new features in 1.1, but the code base has diverged so much, it is practically [...]

Ls3 has so many new features that are not present in main that the list would be pretty long, I think the main thing for most would be the experimental rendering cache.

lagnat, 2025-10-06:
If you're saying that you'd port some of your enhancements back to 1.1, I'm sure there will be plenty of suggestions.  For me, I'd really like to see the re-assign external geometry feature in 1.x.

drwho495, 2025-10-06:
> If you're saying that you'd port some of your enhancements back to 1.1, I'm sure there will be plenty of suggestions.  For me, I'd really like to see the re-assign external geometry feature in 1.x.

We definitely need that aswell

6rzegorz, 2025-10-06:
I have been waiting for two years for the implementation of the **Variable fillet radius** feature, which is only available in the Link branch. I would be grateful if it could be added to 1.1. code base.

prokoudine, 2025-10-06:
> Ls3 has so many new features that are not present in main that the list would be pretty long, I think the main thing for most would be the experimental rendering cache.

That list already exists

https://docs.google.com/document/d/1dDqC5jCO9_eYgMTJEbwpS14rdp_nRI9OkhjgEiLYqtM/edit?tab=t.0

belinski-roman, 2025-10-19:
Please add the auto-hide function of the navigation cube

[image]

This allows you to keep your workspace clean without unnecessary distractions

[image]

[image]

In addition, I really like how you display the cube itself, namely that it is always with perspective distortion, regardless of the mode in the 3d view window. This makes it easier to interact with it

belinski-roman, 2025-10-19:
Please, if possible, transfer these tools to Part Design

[image]

belinski-roman, 2025-10-19:
I would like to see your next improvements inside the Part Design tools (I remember that there were discussions about this on the forum, but I don't remember what they came to). This will reduce the number of buttons on the panel by half.

[image]

In addition, the function of replacing the section for the operation could also be very useful

[image]

belinski-roman, 2025-10-19:
I don'T know how difficult it is to transfer the next feature (I believe that a more recent Coin is needed), but it is also very useful. I'm talking about a section filled with shading.

[image]

6rzegorz, 2025-12-03:
> However, it might be more feasible to pick things you want and add to 1.1 code base. So for everyone still interested, please list the things you like, and I will see if I can add.

[@realthunder](https://github.com/realthunder) Please let us know when we can expect that you implement these important features from Link to main branch. Is it a matter of a month, two, or maybe six months?

belinski-roman, 2026-02-13:
A more extensive list of Draw Style

[image]

[image]