realthunder/FreeCAD #1056: [Problem] How to compile for Windows?

jmozmoz, 2025-01-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
The current LinkStable branch (a9810d509a6f112b5ac03d4d4831b67e6bffd5b7)
```

### Subproject(s) affected?

None

### Problem description

I tried to compile the latest commit of the LinkStable branch (a9810d509a6f112b5ac03d4d4831b67e6bffd5b7) on Windows, but I ran into different problems, depending on the combination of compiler and libpack.

- Visual Studio 2022 and libpack versions below LibPack-1.0.0 Version 3.0.0: The error described here: https://github.com/FreeCAD/FreeCAD/issues/17813
 
- Visual Studio Build Tools 2019 and libpack versions >=3.0.0: Does not work, because QT 5 is missing
 
- Visual Studio Build Tools 2019 and libpack versions LibPack-0.21-V2.11 and LibPack-OCC76-V2 (the default used in GitHub pipelines): Compiler error in Part.dll
```
xxx\FreeCAD\src\Mod\Part\App\AppPartPy.cpp(393,51): error C2664: 'HMODULE LoadLibraryW(LPCWSTR)': cannot convert argument 1 from 'const char [11]' to 'LPCWSTR' [xxx\FreeCAD\build19\src\Mod\Part\App\Part.vcxproj]
xxx\FreeCAD\src\Mod\Part\App\TopoShapeEx.cpp(2293,21): error C2653: 'BRepFeat': is not a class or namespace name [xxx\FreeCAD\build19\src\Mod\Part\App\Part.vcxproj]
xxx\FreeCAD\src\Mod\Part\App\TopoShapeEx.cpp(2293,31): error C3861: 'FaceUntil': identifier not found [xxx\FreeCAD\build19\src\Mod\Part\App\Part.vcxproj]
xxx\FreeCAD\src\Mod\Part\App\TopoShapeEx.cpp(2295,21): error C2653: 'BRepFeat': is not a class or namespace name [xxx\FreeCAD\build19\src\Mod\Part\App\Part.vcxproj]
xxx\FreeCAD\src\Mod\Part\App\TopoShapeEx.cpp(2295,31): error C3861: 'FaceUntil': identifier not found [xxx\FreeCAD\build19\src\Mod\Part\App\Part.vcxproj]
```

What combination of LibPack/compiler is working for Windows?

After `Configure` and `Generate` in `cmake-gui`, I used this command to start the build:
```
cmake --build . --config Release --target ALL_BUILD
```

### Anything else?

When checking out the submodules, one points to a non-existing commit:
```
git submodule update --recursive --init
fatal: remote error: upload-pack: not our ref f0ccbe232f05ef4113143b8cd6c6c307de362aae
fatal: Fetched in submodule path '../src/3rdParty/Diligent/DiligentCore/ThirdParty/xxHash', but it did not contain f0ccbe232f05ef4113143b8cd6c6c307de362aae. Direct fetching of that commit failed.
fatal: Failed to recurse into submodule path '../src/3rdParty/Diligent/DiligentCore'
fatal: Failed to recurse into submodule path '../src/3rdParty/Diligent'
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Shkolik, 2025-02-17:
Same question... Would be great to have if not instruction, but at least version of the lib-pack to build a project.

legion-96, 2025-02-20:
I have been trying to figure this out as well itd be pretty cool if someone could help answer this