realthunder/FreeCAD #1058: Can we have a "Sketcher Offset" tool ported from main branch?

TomaszSzt, 2025-02-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: a9810d509a6f112b5ac03d4d4831b67e6bffd5b7
Python 3.11.8, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * Help 1.0.3
  * InventorLoader 1.4.0
  * sheetmetal 0.4.11
[/code]
```

### Subproject(s) affected?

None

### Problem description

I would be glad to see this feature in LinkBranch:

https://wiki.freecad.org/Sketcher_Offset

Is it possible? Or should I just enable / download some plugin maybe?

It would help me a lot in my projects.

Thanks for an excellent work, I find this branch of FreeCad to be very good and focusing on actual needs of mechanical design.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

wohltat, 2025-02-18:
Other people have also asked for that. 
https://github.com/realthunder/FreeCAD/issues/1055#issuecomment-2628858503

It is not what you asked for but there is an offset tool already. 
Use the green sub-object binder of your sketch, face or part. This generated a copy that has the property category "offsetting". There you can set an offset and the type of offset. 

This is a very powerful tool since it not only works in 2d but also 3d, but sadly not for wires. This is a big since it is possible when using the draft workbench.
In 2d closed you can also use 'offset fill' to only get the offseted border.

3xxx, 2025-02-19:
[image]
green "sub-object(s) shape binder"

6rzegorz, 2025-02-20:
I think Shape binder is not the right tool for offsetting. I am also waiting for appearing the Offset tool in Sketcher.
It would be nice also have possibility to offset 2D shapes and B-splines as well.

TomaszSzt, 2025-02-21:
Thanks. I manged to figure out this work around, but since offset is in main FreeCad I would be happy to see it in this excellent branch too.