realthunder/FreeCAD #1063: [Problem] freecadcmd executed twice,and the saveImage always has a small image

3xxx, 2025-03-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: a9810d509a6f112b5ac03d4d4831b67e6bffd5b7
Python 3.11.8, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: Chinese/China (zh_CN)
Installed mods: 
  * A2plus 0.4.66
  * A2plus.backup1724687131.225395 0.4.64 (Disabled)
  * Assembly3 0.12.2
  * Assembly4 0.50.15
  * Assembly4.backup1724683001.0928686 0.50.9 (Disabled)
  * BillOfMaterials 0.0.7
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

I meet 3 problems:
I have python like:and python file name is "Testtest.py"
```
# FREECADPATH = r'D:/FreeCAD_weekly-builds-39980-conda-Windows-x86_64-py311/bin' # master
# FREECADPATH = r'D:/FreeCAD-Link-Stable-Win-x86_64-py3.11-20241003/bin' # LINK
import sys
if not FREECADPATH in sys.path:
    # if DBG_LOAD is True:
        # print("no module path")
    sys.path.insert(1,FREECADPATH)

import argparse
import FreeCAD
import FreeCADGui as Gui
import Part # , Mesh
import time

Gui.showMainWindow()
window = Gui.getMainWindow()
# window.showMinimized()
import ImportGui

if __name__ == '__main__':
    print(len(sys.argv))
    print(sys.argv[1])
    print(sys.argv[2])
    print(sys.argv[3])
    inputfile = sys.argv[2]
    outputfile = sys.argv[3]
    doc = FreeCAD.openDocument(inputfile)
    App.setActiveDocument(doc.Name)
    App.ActiveDocument=App.getDocument(doc.Name)
    Gui.ActiveDocument=Gui.getDocument(doc.Name)
    time.sleep(5)
    Gui.activeDocument().activeView().saveImage(outputfile,800,600,'Current')
    time.sleep(5)
```
1.problem 1：when run it in  freecadmaster freecadcmd，it's ok
```
D:\FreeCAD_weekly-builds-39980-conda-Windows-x86_64-py311\bin>freecadcmd testtest.py d:\333.fcstd d:\333.jpg
FreeCAD 1.1.0, Libs: 1.1.0devR39980 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

NOTICE: Addon "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\A2plus.backup1724687131.225395" disabled by presence of ADDON_DISABLED stopfile
NOTICE: Addon "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\Assembly4.backup1724683001.0928686" disabled by presence of ADDON_DISABLED stopfile
NOTICE: Addon "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\A2plus.backup1724687131.225395" disabled by presence of ADDON_DISABLED stopfile
NOTICE: Addon "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\Assembly4.backup1724683001.0928686" disabled by presence of ADDON_DISABLED stopfile
Main window restored
Show main window
Toolbars restored
4
testtest.py
d:\333.fcstd
d:\333.jpg
d:\333.fcstd
<Spreadsheet> Cell.cpp(676): _33#Spreadsheet.E1: Unknown function
Not detached all observers yet
Not detached all observers yet
```
when run in freecad link freecadcmd like this: it run twice
```
D:\FreeCAD-Link-Stable-Win-x86_64-py3.11-20241003\bin>freecadcmd testtest.py d:\2222.fcstd d:\20250308.jpg
FreeCAD 0.21.0, Libs: 2024.1006.0R14555 (Git shallow)
NOTICE: Addon "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\A2plus.backup1724687131.225395" disabled by presence of ADDON_DISABLED stopfile
NOTICE: Addon "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\Assembly4.backup1724683001.0928686" disabled by presence of ADDON_DISABLED stopfile
QObject::connect: No such signal QApplication::messageReceived(const QList<QByteArray> &)
QObject::connect: No such signal QApplication::messageReceived(const QList<QByteArray> &)
NOTICE: Addon "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\A2plus.backup1724687131.225395" disabled by presence of ADDON_DISABLED stopfile
NOTICE: Addon "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\Assembly4.backup1724683001.0928686" disabled by presence of ADDON_DISABLED stopfile
Main window restored
Show main window
Toolbars restored
4
testtest.py
d:\2222.fcstd
d:\20250308.jpg
4
testtest.py
d:\2222.fcstd
d:\20250308.jpg
Not detached all observers yet
Not detached all observers yet
```
problem2: the image which it save like this: image always has a small image in the middle.

[image]

problem3:The python file name must be uppercase, and the command line must be lowercase, otherwise, if name is "testtest.py", the python code execution will not be complete
```
D:\FreeCAD-Link-Stable-Win-x86_64-py3.11-20241003\bin>freecadcmd testtest.py d:\2222.fcstd d:\20250308.jpg
FreeCAD 0.21.0, Libs: 2024.1006.0R14555 (Git shallow)
NOTICE: Addon "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\A2plus.backup1724687131.225395" disabled by presence of ADDON_DISABLED stopfile
NOTICE: Addon "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\Assembly4.backup1724683001.0928686" disabled by presence of ADDON_DISABLED stopfile
QObject::connect: No such signal QApplication::messageReceived(const QList<QByteArray> &)
QObject::connect: No such signal QApplication::messageReceived(const QList<QByteArray> &)
NOTICE: Addon "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\A2plus.backup1724687131.225395" disabled by presence of ADDON_DISABLED stopfile
NOTICE: Addon "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\Assembly4.backup1724683001.0928686" disabled by presence of ADDON_DISABLED stopfile
Main window restored
Show main window
Toolbars restored
Not detached all observers yet
Not detached all observers yet
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

realthunder, 2025-04-12:
You can't easily run `FreeCADCmd` the same way as standard Python interpreter. It has very different way of parsing the input arguments. There is a standard Python interpreter in each release. Just `cd` to the `bin` directory and run `python.exe` with those arguments and it will work.

3xxx, 2025-04-18:
Yes,it is the right way.

closed 2025-04-18 by 3xxx