realthunder/FreeCAD #1065: [Problem] Pad not recognizing sketcher internal, even though the internal face name is the same.

drwho495, 2025-03-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 2025.316.0dev.14555 (Git shallow) AppImage
Build type: Release
Branch: main
Hash: eec8cdadf00325436c54c8d04c985305fc4afaf7
Python 3.11.8, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Render 2024.12.15 (Disabled)
  * ArchTextures
  * Assembly3 0.12.2
  * CadbaseLibrary 2.0.0
  * Behave-Dark-Colors 0.1.1
  * SearchBar 1.0.2
  * WebTools 1.0.0
  * Manipulator 1.5.7
  * POV-Ray-Rendering
  * cadquery_module 2.1.0
  * Ondsel-Lens 2024.11.29.01 (Disabled)
  * freecad.gears 1.3.0
  * ExplodedAssembly
  * fasteners 0.5.37
  * cadquery_module.backup1741030188.0603402 (Disabled)
  * FreeCAD-Telemetry 1.0.0beta
  * QuickMeasure 2022.10.28
  * FreeCAD-Ribbon 1.8.0
  * assembly2
  * Freecad-Built-in-themes-beta 1.2.2
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

PartDesign

### Problem description

Sometimes, when I create a pad from an internal face of a sketch and then change the face, the pad loses reference to the face, even though the face name and ID are the exact same.

When trying to record the video, I wasn't able to get the issue to repeat the first few times, so it is intermittent.

https://github.com/user-attachments/assets/d6b01d97-ba29-4200-a910-a44d9dc42b83

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

wohltat, 2025-03-17:
I could replicate that phenomenon.
When i stat FC from the terminal then i get the following error there when the sketch is changed (Sketch still has only one face):

```
9.7e-08 <App> Document.cpp(3461): Failed to recompute Unnamed#Pad: Sub shape not found: Unnamed#Sketch.?InternalFace1
Recompute failed!
```

wohltat, 2025-03-17:
Here are the files to a minimal example of the bug:
[internal_face_broken_mapping.zip](https://github.com/user-attachments/files/19299479/internal_face_broken_mapping.zip)

There is a file "before" and "after" the bug occurs.

This little script is enough to fix the InternalFace reference in the profile, in the `...after.FCStd`.

``` python
profile = App.ActiveDocument.Pad.Profile

sketch = profile[0]
internal_face_list = profile[1]

face = internal_face_list[0]
face = face.lstrip('?')
App.ActiveDocument.Pad.Profile = (sketch, [face])
App.ActiveDocument.recompute()
```
This does not say why it gets broken, but shows that it just gets falsely marked invalid.

wohltat, 2025-03-20:
Here is a little addition to the script that you can put in a macro and run as a workaround.
It will restore the previous reference of any pad in the document expecting it exists:

``` python
doc = App.ActiveDocument
pads = [obj for obj in doc.Objects if obj.TypeId == "PartDesign::Pad"]

for pad in pads:
    (sketch, internal_face_list) = pad.Profile

    fixed_face_list  = [face.lstrip('?') for face in internal_face_list]
    pad.Profile = (sketch, fixed_face_list)
doc.recompute()
```

drwho495, 2025-03-22:
@realthunder

realthunder, 2025-04-16:
The above commit can only solve for cases where there is exactly one sub-shape type in the new shape, which is what's shown in your case.

The reason of the `intermittent` failure is because, the face topo name only uses one edge for naming. Therefore, the face topo name stays the same unless you replace that edge. The above commit can resolve the missing name if there is only one face to choose in the new shape. It will still fail if there are more. However, you can avoid this kind of topo naming problem by using the `Swap geometry ID` command. To replace an edge, DO NOT delete the old edge. Instead, create a new edge between the vertices of the old edge. Then select those two edges and invoke the `Swap geometry ID` command. After that, you can delete the old edge. Or, make it construction in case you want to go back.

[image]

drwho495, 2025-06-19:
> The above commit can only solve for cases where there is exactly one sub-shape type in the new shape, which is what's shown in your case.
>
> The reason of the `intermittent` failure is because, the [...]

would there be a permanent fix for this that could allow the elementmap to reference more than one sketch geo id?