realthunder/FreeCAD #1067: [Problem]  combo view elements disappearing when losing focus

wohltat, 2025-03-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Manjaro Linux (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 2025.316.0dev.14555 (Git shallow) AppImage
Build type: Release
Branch: main
Hash: eec8cdadf00325436c54c8d04c985305fc4afaf7
Python 3.11.8, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

PartDesign

### Problem description

First i want to say i'm really exited how this is going and really love to now see the new features being merged. 

Parts of the combo view hide away in overlay mode when the focus is lost. This  does not happen in non-overlay-mode.
Happens for any of the transparency and auto-hide settings.

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

realthunder, 2025-03-20:
I can't reproduce the behavior. Could you please attach your user conf file here? Tools -> Edit parameters -> Export.

wohltat, 2025-03-20:
Hi,
here is the config exported from the tip-version:

[combo_view_glitch_1067.FCParam.zip](https://github.com/user-attachments/files/19368091/combo_view_glitch_1067.FCParam.zip)

wohltat, 2025-03-20:
> I can't reproduce the behavior.

That's strange. I just tried with a fresh start (deleted all config files). 
I tried with default style sheets, as well as with other style sheets.
I always get that behavior when using overlay mode with transparency.

commit a6524718 (2025-03-23): Gui: fix overlay tab bar visibility change

Fixes #1067