realthunder/FreeCAD #1077: [FR] Store config in own FreeCadLink folder

vovodroid, 2025-05-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
2025.03.11
```

### Subproject(s) affected?

None

### Problem description

Currently configuration is shared between main project and this one, thing that could create mess in some situations.
I suggest store it C:\Users\User\AppData\Roaming\FreeCADLInk folder.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

realthunder, 2025-05-06:
The release image uses `branding.xml` file to add a prefix to differentiate the config file, that is, `link.user.cfg`. Are you compiling from the source? If so, download a release image, extract the file and search for `branding.xml`, and put it in the same relative location in your build directory.

vovodroid, 2025-05-06:
> Are you compiling from the source?

No, releases from GitHub. `branding.xml` contains `<VendorCFGPrefix>link.</VendorCFGPrefix>`