realthunder/FreeCAD #1078: [Problem] 3mf export breaks import in Slicer

wohltat, 2025-05-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Manjaro Linux (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: a9810d509a6f112b5ac03d4d4831b67e6bffd5b7
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Launcher
  * ShortCuts
[/code]
```

### Subproject(s) affected?

File formats

### Problem description

There is an inconsistency when exporting .3mf files. 
The export seems to go fine but when importing in a slicer like Bambuslicer or orcaslicer, it fails with an error message:
"The file does not contain any geometry data"

I attached a file where it happens but couldn't figure out yet why or when exactly it happens. But this is how far i got:

I described the problem here already:
https://github.com/bambulab/BambuStudio/issues/3012#issuecomment-2848694701

The problem is in the `3dmodel.model` file inside the 3mf package.

The working export looks like this:
```
<?xml version="1.0" encoding="UTF-8"?>
<model unit="millimeter" xml:lang="en-US" xmlns="http://schemas.microsoft.com/3dmanufacturing/core/2015/02">
 <metadata name="Application">FreeCAD</metadata>
 <resources>
  <object id="1" type="model">
...
```
and the broken export looks like this:
```
...
<object id="1" type="surface">
...
```
It seems that it should always export as `model` and not as `surface`.

### Anything else?

[cam_shaft.FCStd.zip](https://github.com/user-attachments/files/20024180/cam_shaft.FCStd.zip)
works on export of early features like "gear guide 1",
but fails on late features like "center hole".

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

shampeon, 2025-08-28:
I just hit this error in OrcaSlicer 2.3.0 (macOS x64). I had exported a model previously with no problems, but adding a Fillet triggered a "The file does not contain any geometry data." warning in OrcaSlicer. Exporting as an STL file also caused the same error.

```
OS: KDE Flatpak runtime (pop:GNOME/pop/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 256fc7e)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Curves 0.6.70
```

shampeon, 2025-08-28:
Same behavior with 1.1.0 `dev`.

```
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop/xcb)
Architecture: x86_64
Version: 1.1.0dev.43087 (Git)
Build date: 2025/08/26 09:00:30
Build type: Release
Branch: main
Hash: b7d2913b0c7c29c2f0ed99d7d273c5c4d3837d53
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.2
shiboken 5.15.2, SMESH 7.7.1.0, xerces-c 3.2.3, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 120/106.813/1
Installed mods: 
  * Gridfinity 0.11.9
  * FrameForge 0.1.3
  * FeedsAndSpeeds 0.6.0
  * fasteners 0.5.41
  * kicadStepUpMod 11.7.3
  * btl 0.9.9
  * Marz 0.1.16
  * WB_Organizer 2024.1.29
```